tinyfpga/TinyFPGA-Bootloader #2: Port to Xilinx Series-7 (such as the Artix) FPGAs

mithro, 2018-02-12:
It would be awesome if this bootloader worked on Xilinx Series 7 FPGAs.

There is information on how to do multiboot on these devices in the following pages;
 * [XAPP1247 (v1.1) February 28, 2017 MultiBoot with 7 Series FPGAs and SPI](https://www.xilinx.com/support/documentation/application_notes/xapp1247-multiboot-spi.pdf)
 * [AC701 MultiBoot Design](https://www.xilinx.com/support/documentation/boards_and_kits/ac701/2014_3/xtp226-ac701-multiboot-c-2014-3.pdf)
 * [KC705 MultiBoot Design](https://www.xilinx.com/support/documentation/boards_and_kits/kc705_multiboot_pdf_xtp104_14.2.pdf)

tinyfpga, 2018-05-17:
> @mithro <https://github.com/mithro> I caught talks with you guys at
> teardown over the weekend. I'm forking with the intention of adding vivado
> build support and testing the usb side on a parallella (z7010) because that
> is what I have. Once this is working I'll look at the flash interface and
> multiboot. I'm a vhdl guy so I won't be quick.

I have USB and SPI portions working on the arty, but without multiboot so
far.  I’ll push my changes as soon as I can so you can see what it takes.

TomKeddie, 2018-05-17:
"IPROG Reconfiguration" in chapter 7 of [UG470](https://www.xilinx.com/support/documentation/user_guides/ug470_7Series_Config.pdf) seems to have the necessary information to softboot the device from a different address in flash.  I don't have access to a 7 series board with flash, I'll see what I can find.

TomKeddie, 2018-06-15:
> I have USB and SPI portions working on the arty, but without multiboot so
far.  I’ll push my changes as soon as I can so you can see what it takes.
@tinyfpga I realise you're super busy with the BX.  I have an arty7-35 and have this code building in Vivado.  When you have a chance, pls push your code and I'll take a look at adding the reconfig.  Thanks, Tom.

tinyfpga, 2018-06-15:
@TomKeddie: take a look here: https://github.com/tinyfpga/TinyFPGA-Bootloader/tree/master/boards/Digilent-Arty

TomKeddie, 2018-06-20:
I created a pmod pcb to work with this code it seems to work well.  See https://github.com/TomKeddie/tom-prj-pmod/tree/master/hardware/pmod-usb-raw.

If you want one I'd be happy to send you one.

If it's ok with you I'd like to migrate the pinout to use one of the pmod connectors, people can either get some pcbs made ($2 at oshpark) or use a breakout board with the pmod pins.

tinyfpga, 2018-06-20:
That’s awesome! I’ll accept a pull-request to remap the pins to your pmod.
I suggest adding the BOM on octopart as well.  It makes it easy to order
all the parts.

closed 2019-07-06 by mithro