tinyfpga/TinyFPGA-Bootloader #5: UP5K with internal osc breaks..

aventuri, 2018-02-28:
hi Luke, the "poor man" USB bootloader concept is a VERY clever idea! thanx.. :-)

i've just tried to "compile" the Icebreaker-bitsy flavour with just a little tweak for using the internal OSC (to test eventually on UPDuino,, [*]).

`ifdef INT_CLK
  wire pin_clk;
        // try with the internal clock, i've put the VCCPLL caps.. 
   SB_HFOSC OSCInst0(
                     .CLKHFEN(1'b1),
                     .CLKHFPU(1'b1),
                     .CLKHF(pin_clk)   );
   defparam OSCInst0.CLKHF_DIV = "0b00";  // 48MHz DIVIDED by 1
`endif

..but the place&route step [breaks this way](https://paste.debian.net/1012332/)

BTW this log is about when i tried to skip also the instantiation of the PLL, but the break-up is the same..

is this report triggering any hint about solution?

thanx

andrea

[*] i know the UPDuino is a very broken design, in term of signal integrity! i just added small caps for improving PLL, but anyway, i know it would be more difficult for USB to works, then other board.
the issue on the "firmware" routing is anyway independent. and if those Lattice guys put a 48MHz int OSC, it means probably that the USB interfaces was a "use case" to be considered IMHO.

daveshah1, 2018-02-28:
This seems to be an arachne-pnr issues, which I'll look into - it seems that its promoting a global net onto the global used for the HFOSC output. I'm not sure why that is, I fixed an issue like that a while back and haven't seen it since but this seems to be a case not covered by that fix.

In the mean time, put the following attribute immediately before the SB_HFOSC as a workaround:
```
(* ROUTE_THROUGH_FABRIC=1 *)
```

aventuri, 2018-02-28:
yeah, thanx Dave for the quick workaround.

now i can create the bootloader.asc -> and then the .bin with icepak.

icetime still bumps this way:

```
icetime -d up5k -mtr bootloader.rpt bootloader.asc
// Reading input .asc file..
// Reading 5k chipdb file..
// Creating timing netlist..
terminate called after throwing an instance of 'std::invalid_argument'
  what():  stoi
Makefile:40: recipe for target 'bootloader.rpt' failed
make: *** [bootloader.rpt] Aborted (core dumped)
make: *** Deleting file 'bootloader.rpt'

```

aventuri, 2018-02-28:
just a followup about "the real test". short summary: still no joy!

here the[ connection](https://www.youtube.com/watch?v=MHc6dZ-8OQg).

here the dmesg excerpt:
```
[127313.191670] usb 1-2.4.2: new full-speed USB device number 27 using xhci_hcd
[127313.275707] usb 1-2.4.2: device descriptor read/64, error -32
[127313.483730] usb 1-2.4.2: device descriptor read/64, error -32
[127313.691685] usb 1-2.4.2: new full-speed USB device number 28 using xhci_hcd
[127313.791743] usb 1-2.4.2: device descriptor read/64, error -32
[127313.999741] usb 1-2.4.2: device descriptor read/64, error -32
[127314.207688] usb 1-2.4.2: new full-speed USB device number 29 using xhci_hcd
[127314.207778] usb 1-2.4.2: Device not responding to setup address.
[127314.415802] usb 1-2.4.2: Device not responding to setup address.
[127314.623702] usb 1-2.4.2: device not accepting address 29, error -71
[127314.723719] usb 1-2.4.2: new full-speed USB device number 30 using xhci_hcd
[127314.723818] usb 1-2.4.2: Device not responding to setup address.
[127314.931788] usb 1-2.4.2: Device not responding to setup address.
```
next move, as i have an UART handy, i plan to setup some kind of instrumentation  for it (at least to check the internal clk is really 48MHz..)  to deal with printing out the internal state machine and check where it hangs. of course USB is a HUUGE spec, so it's a bit of a shot in the dark.

anyway i suppose this "issue" about FW creation could be closed now.

daveshah1, 2018-02-28:
I think @esden is also working on this, you might want to talk to him.

Please could you put your exact code, constraints and Makefile somewhere, so I can look into the arachne and icetime issues?

aventuri, 2018-02-28:
you can find the modded files in Upduino dir on this repo/branch:
https://github.com/aventuri/TinyFPGA-Bootloader/tree/20180228_intOsc

i'd like to give you the 'icetime' version but i miss to see the param that could print out the "git commit" that sourced it .. could be an useful addendum, as there's still some development ongoing, in the foreseable future..

aventuri, 2018-02-28:
...my icetime binary has been compiled from this source code:
```
commit 1d84b7a53183df3d2d620d5633ca82db6ab90186 (HEAD -> master, origin/master, origin/HEAD)
Merge: edbf5fc 58e2e74
Author: Clifford Wolf <clifford@clifford.at>
Date:   Wed Feb 14 18:48:09 2018 +0100

    Merge pull request #124 from daveshah1/ultra
    
    Add iCE40 Ultra (ice5lp) support to icepack
[andrea@CHUWI123 ~/fpga/icestorm/icetime (master)] $ 

```

daveshah1, 2018-02-28:
Thanks for sharing your code.

The icetime issue has been fixed in icestorm PR #129 (I hadn't tested the use of route_through_fabric with icetime before), but beware that icetime is still not giving a useful result because there are logic loops in the design (that is a bigger issue, and as much of a problem with the design as icetime IMO).

The arachne-pnr issue still needs investigation.

tinyfpga, 2018-07-26:
I don't believe that the UP5K internal oscillator is stable enough or accurate enough to be used as a source clock for USB.

closed 2018-07-26 by tinyfpga