tinyfpga/TinyFPGA-Bootloader #12: "Fixing TinyFPGA BX board with wrong USB VID:PID" crashes after erasing bootloader

kevinmehall, 2018-07-12:
I installed tinyprog 1.0.5 with pip on Ubuntu 18.04 and ran it for the first time and got this:
```
% tinyprog
Fixing TinyFPGA BX board with wrong USB VID:PID...
    Erasing:   0%|                                                                            | 0.00/299k [00:00<?, ?B/s]
Traceback (most recent call last):
  File "/home/km/.local/bin/tinyprog", line 11, in <module>
    sys.exit(main())
  File "/home/km/.local/lib/python2.7/site-packages/tinyprog/__main__.py", line 45, in main
    check_for_wrong_tinyfpga_bx_vidpid()
  File "/home/km/.local/lib/python2.7/site-packages/tinyprog/__main__.py", line 29, in check_for_wrong_tinyfpga_bx_vidpid
    if p.program_bitstream(0, tinyfpga_bx_fw_bitstream):
  File "/home/km/.local/lib/python2.7/site-packages/tinyprog/__init__.py", line 344, in program_bitstream
    return self.program(addr, bitstream)
  File "/home/km/.local/lib/python2.7/site-packages/tinyprog/__init__.py", line 303, in program
    self.erase(addr, len(data), disable_progress=False)
  File "/home/km/.local/lib/python2.7/site-packages/tinyprog/__init__.py", line 272, in erase
    self._erase(addr, erase_length)
  File "/home/km/.local/lib/python2.7/site-packages/tinyprog/__init__.py", line 225, in _erase
    self.wait_while_busy()
  File "/home/km/.local/lib/python2.7/site-packages/tinyprog/__init__.py", line 213, in wait_while_busy
    while ord(self.read_sts()) & 1:
TypeError: ord() expected a character, but string of length 0 found
```

and well, it definitely doesn't have the wrong PID anymore, but doesn't seem to have the right one, or a bootloader at all for that matter...

```
[610009.513632] usb 2-1: new full-speed USB device number 47 using xhci_hcd
[610009.641669] usb 2-1: device descriptor read/64, error -71
[610009.877713] usb 2-1: device descriptor read/64, error -71
[610010.113683] usb 2-1: new full-speed USB device number 48 using xhci_hcd
[610010.241707] usb 2-1: device descriptor read/64, error -71
[610010.477722] usb 2-1: device descriptor read/64, error -71
[610010.589720] usb usb2-port1: attempt power cycle
[610011.241630] usb 2-1: new full-speed USB device number 49 using xhci_hcd
[610011.241844] usb 2-1: Device not responding to setup address.
[610011.449930] usb 2-1: Device not responding to setup address.
[610011.657654] usb 2-1: device not accepting address 49, error -71
[610011.785723] usb 2-1: new full-speed USB device number 50 using xhci_hcd
[610011.785929] usb 2-1: Device not responding to setup address.
[610011.993873] usb 2-1: Device not responding to setup address.
[610012.201740] usb 2-1: device not accepting address 50, error -71
[610012.201823] usb usb2-port1: unable to enumerate USB device
```

kevinmehall, 2018-07-12:
Recovered my board by flashing https://github.com/tinyfpga/TinyFPGA-Bootloader/releases/download/1.0.1/tinyfpga_bx_fw.bin at address 0x0:0x48fbc with a SPI flash programmer.

kevinmehall, 2018-07-12:
Lenovo T440s

```
00:00.0 Host bridge: Intel Corporation Haswell-ULT DRAM Controller (rev 0b)
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 0b)
00:14.0 USB controller: Intel Corporation 8 Series USB xHCI HC (rev 04)
00:16.0 Communication controller: Intel Corporation 8 Series HECI #0 (rev 04)
00:16.3 Serial controller: Intel Corporation 8 Series HECI KT (rev 04)
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection I218-LM (rev 04)
00:1b.0 Audio device: Intel Corporation 8 Series HD Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 6 (rev e4)
00:1c.1 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 3 (rev e4)
00:1d.0 USB controller: Intel Corporation 8 Series USB EHCI #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation 8 Series LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 8 Series SATA Controller 1 [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 8 Series SMBus Controller (rev 04)
02:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5227 PCI Express Card Reader (rev 01)
03:00.0 Network controller: Intel Corporation Wireless 7260 (rev 83)
```
I also dumped the TinyFPGA flash before rewriting it, and can confirm that it was erased to 0xFF from address 0 to 0x0010000.

After `tinyprog` failed with that traceback, I instinctively power-cycled the USB hub before noticing that it had been attempting to flash the bootloader. I wonder if the FPGA was still in a state that another run could have recovered from more easily given that the bitstream was already in configuration RAM. Would have been more careful if I had known it was doing that.

tinyfpga, 2018-07-12:
It may or may not have been in a healthy state at that point in time.  Sometimes the USB serial driver acts funny.  I'm backing off of the automatic bootloader update.  The 1.0.7 release will accept both the old and new VID:PID.  In the near future I'll add the bootloader update commandline option with appropriate warnings.  The idea will be to program the bootloader to the user area first, boot to the new bootloader, then overwrite the old bootloader.

tinyfpga, 2018-07-12:
I've disabled the automatic upgrade and modified the following line to use either the old or new VID:PID: https://github.com/tinyfpga/TinyFPGA-Bootloader/blob/3ee049ce9aff8cdecaaac95670684d9de2398bde/programmer/tinyprog/__main__.py#L83

closed 2018-07-12 by tinyfpga