tinyfpga/TinyFPGA-Bootloader #18: Error while updating bootloader:  invalid literal for int() with base 10

shenki, 2018-08-01:
I have installed tinyprog 20585978638699349b7739b4eec80f6703f969b8 using `python setup.py install --user`, an was using it to update the bootloader my TinyFPGA BX:
```
$ tinyprog --update-bootloader

    TinyProg CLI
    ------------
    Using device id 1d50:6130
    Only one board with active bootloader, using it.

    The following update:

        New Version: 1.0.1
        Notes: Updates USB VID:PID to fix issues with APIO not recognizing the board.

    is available for this board:

        /dev/ttyACM0: TinyFPGA BX 1.0.0
            UUID:
            FPGA: ice40lp8k-cm81

    Would you like to perform the update? [y/N] y
    Fetching stage one...
    Programming stage one...
    Erasing: 100%|████████████████████████████████████████████████████████████████████████████████████████████| 135k/135k [00:00<00:00, 163kB/s]
    Writing: 100%|████████████████████████████████████████████████████████████████████████████████████████████| 135k/135k [00:00<00:00, 177kB/s]
    Reading: 100%|████████████████████████████████████████████████████████████████████████████████████████████| 135k/135k [00:00<00:00, 425kB/s]
    ...Success!
    Waiting for stage one to reconnect....connected!
    Fetching stage two...
    Programming stage two...
    Erasing:   0%|                                                                                                   | 0.00/299k [00:00<?, ?B/s]
Traceback (most recent call last):
  File "/home/joel/.local/bin/tinyprog", line 11, in <module>
    load_entry_point('tinyprog==1.0.18', 'console_scripts', 'tinyprog')()
  File "build/bdist.linux-x86_64/egg/tinyprog/__main__.py", line 282, in main
  File "build/bdist.linux-x86_64/egg/tinyprog/__main__.py", line 152, in perform_bootloader_update
  File "build/bdist.linux-x86_64/egg/tinyprog/__init__.py", line 419, in program_bitstream
  File "build/bdist.linux-x86_64/egg/tinyprog/__init__.py", line 375, in program
  File "build/bdist.linux-x86_64/egg/tinyprog/__init__.py", line 347, in erase
  File "build/bdist.linux-x86_64/egg/tinyprog/__init__.py", line 301, in _erase
  File "build/bdist.linux-x86_64/egg/tinyprog/__init__.py", line 290, in wait_while_busy
  File "build/bdist.linux-x86_64/egg/tinyprog/__init__.py", line 21, in to_int
ValueError: invalid literal for int() with base 10: ''
```

ewenmcneill, 2019-05-17:
For the benefit of anyone else finding this, @shenki also posted on the TinyFPGA Discourse forum, and got this suggestion on how it might be possible to fix with a soldering iron and/or a SPI flash programmer:

https://discourse.tinyfpga.com/t/error-while-updating-bootloader-invalid-literal-for-int-with-base-10/484/3

(See also https://github.com/tinyfpga/TinyFPGA-Bootloader/issues/43 for another user with similar symptoms.)

Ewen

paulo-raca, 2019-07-05:
I had exactly the same problem (Twice! I bricked my 2 boards :man_facepalming: )

I'll try the arduino programmer, but is there anything in finding the reason/fixing the bug?

ewenmcneill, 2019-07-06:
@paulo-raca Sorry to hear that you had the same problem.  Hope you're able to recover them (there are a few suggestions on the TinyFPGA forum for various ways to reflash the initial bootloader externally).

I don't think anyone has figured out *exactly* what causes the problems (to me it feels like a timing race between the `tinyprog` serial software, the TinyFPGA bootloader verilog, and the SPI flash chip used).  There's a couple of attempts at working around it (eg, these pull requests that haven't been merged -- https://github.com/tinyfpga/TinyFPGA-Bootloader/pull/52, https://github.com/tinyfpga/TinyFPGA-Bootloader/pull/16), but my one (https://github.com/tinyfpga/TinyFPGA-Bootloader/pull/52) didn't seem to be sufficient to fix everything by itself and the other one earlier one (https://github.com/tinyfpga/TinyFPGA-Bootloader/pull/16) contains lots of other things so it's not clear it'll ever get merged.

So far the best work around seems to be:

*    Longer serial timeouts (in my pull request)

*    Write full "tiny sectors" (256 octets) as the issue seems easier to trigger on certain part-sector sizes (in my pull request)

*    Retry the failed sector writes a few times (in https://github.com/tinyfpga/TinyFPGA-Bootloader/pull/16)

Unfortunately the original TinyFPGA developer seems to have had no time to do anything with this in the last year (all the more recent commits are others, often written or merged by Tim (mithro) who is also busy).  So it's not clear the underlying cause will ever be debugged fully.  But perhaps it'll eventually be possible to get enough workarounds (eg, all *three* of the above) merged, and a new full release made (there's only been dev releases in the last year, which have to be explicitly installed).

(It'd also help if the TinyFPGA boards shipped with the newest bootloader on them, but manufacturing/delivery pipelines tend to be relatively long, so I'm not sure if a new batch has been made in the last year....)

Ewen

mithro, 2019-07-06:
I believe this has been fixed now?

@ewenmcneill Is that your understanding?

closed 2019-07-06 by mithro

ewenmcneill, 2019-07-06:
> I believe this has been fixed now? 
> @ewenmcneill Is that your understanding?

It is hopefully less likely to occur now with https://github.com/tinyfpga/TinyFPGA-Bootloader/pull/52 (and the symptoms will have changed if it does, as the `invalid literal for int()` was in the error handling code which has been patched separately already).  I suspect we probably need https://github.com/tinyfpga/TinyFPGA-Bootloader/issues/55 as well for a more complete solution (and/or someone debugging how the state machine gets stuck -- it feels like a race to me).

But yes, at this stage it seems useful to consolidate down to one open issue for the "bootloader reprogramming timed out" problem.

Ewen