tinyfpga/TinyFPGA-Bootloader #23: Programming TinyFPGA BX with userdata of size exactly 1244 bytes fails

matt-kimball, 2018-09-16:
When I attempt to program with a PicoSoC hardware and a firmware of size exactly equal to 1244, the programming step fails when reading back the user data:

```
    TinyProg CLI
    ------------
    Using device id 1d50:6130
    Only one board with active bootloader, using it.
    Programming /dev/ttyACM0 with /tmp/hardware.bin
    Programming at addr 028000   
    Waking up SPI flash
    135100 bytes to program
    Erasing: 100%|████████████████████████████| 135k/135k [00:01<00:00, 112kB/s]
    Writing: 100%|███████████████████████████| 135k/135k [00:02<00:00, 62.3kB/s]
    Reading: 100%|████████████████████████████| 135k/135k [00:01<00:00, 111kB/s]
    Success!
    Programming /dev/ttyACM0 with ./firmware.1244.bin
    Programming at addr 050000   
    Waking up SPI flash
    1244 bytes to program
    Erasing: 100%|█████████████████████████| 1.24k/1.24k [00:00<00:00, 11.5kB/s]
    Writing: 100%|█████████████████████████| 1.24k/1.24k [00:00<00:00, 39.2kB/s]
    Reading: 100%|█████████████████████████| 1.24k/1.24k [00:01<00:00, 1.23kB/s]
    Failure!

```

Programming with firmware size of 1240 succeeds, and programming with firmware size of 1248 also succeeds.  And yet, size 1244 fails.  The firmware contents don't seem to matter -- a file generated with dd to have this specific size also fails:

```
dd if=/dev/zero of=firmware.1244.bin bs=1244 count=1
```

mithro, 2019-07-06:
@ewenmcneill - I believe this has been fixed with your changes right?

closed 2019-07-06 by mithro

ewenmcneill, 2019-07-06:
> @ewenmcneill - I believe this has been fixed with your changes right?

I haven't tested with a 1244 byte firmware file, but yes since 1244 = 4 * 256 + 220, which https://github.com/tinyfpga/TinyFPGA-Bootloader/pull/52 would expand out to 5 * 256 (with 0xff fill at the end) it will hopefully now work.

Ewen