tinyfpga/TinyFPGA-Bootloader #27: Latest version attempting to brick Ex prototypes

daveshah1, 2018-09-30:
The latest version of the programmer prints the following message for an ECP5-based Ex:
```

    !!! WARNING !!!

    The address given may overwrite the USB bootloader. Without the
    USB bootloader the board will no longer be able to be programmed
    over the USB interface. Without the USB bootloader, the board can
    only be programmed via the SPI flash interface on the bottom of
    the board

    Are you sure you want to continue? Type in 'yes' to overwrite bootloader. [yes/NO] > 
```
with no address specified.

Proceeding does indeed brick the board.

mithro, 2019-07-06:
@daveshah1 Do you know if this is still happening?

mithro, 2019-07-06:
@daveshah1 Please reopen if it is.

closed 2019-07-06 by mithro