tinyfpga/TinyFPGA-Bootloader #28: TinyFPGA Bootloader unusable on my desktop

daveshah1, 2018-09-30:
At least with the Ex boards, tinyprog is totally unusable on my desktop, it has never succeeded programming but will fail somewhere with a write timeout.

```
Traceback (most recent call last):
  File "/usr/bin/tinyprog", line 11, in <module>
    load_entry_point('tinyprog==1.0.19', 'console_scripts', 'tinyprog')()
  File "/usr/lib/python3.7/site-packages/tinyprog-1.0.19-py3.7.egg/tinyprog/__main__.py", line 341, in main
  File "/usr/lib/python3.7/site-packages/tinyprog-1.0.19-py3.7.egg/tinyprog/__init__.py", line 419, in program_bitstream
  File "/usr/lib/python3.7/site-packages/tinyprog-1.0.19-py3.7.egg/tinyprog/__init__.py", line 375, in program
  File "/usr/lib/python3.7/site-packages/tinyprog-1.0.19-py3.7.egg/tinyprog/__init__.py", line 347, in erase
  File "/usr/lib/python3.7/site-packages/tinyprog-1.0.19-py3.7.egg/tinyprog/__init__.py", line 301, in _erase
  File "/usr/lib/python3.7/site-packages/tinyprog-1.0.19-py3.7.egg/tinyprog/__init__.py", line 290, in wait_while_busy
  File "/usr/lib/python3.7/site-packages/tinyprog-1.0.19-py3.7.egg/tinyprog/__init__.py", line 256, in read_sts
  File "/usr/lib/python3.7/site-packages/tinyprog-1.0.19-py3.7.egg/tinyprog/__init__.py", line 242, in cmd
  File "/usr/lib/python3.7/site-packages/tinyprog-1.0.19-py3.7.egg/tinyprog/__init__.py", line 66, in write
  File "/usr/lib/python3.7/site-packages/serial/serialposix.py", line 579, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout
```

This is i5-4690K CPU, Z97 chipset, running Arch Linux (kernel 4.18.10-arch1-1-ARCH). I have tried front and back USB ports, with and without a hub.

For reference, on my laptop (i7-6700HQ), it succeeds about 50% of the time.

KyKyPi, 2019-02-14:
Hey, I'm running into the exact same error. Did you ever solve your problem?

mithro, 2019-07-06:
@daveshah1 Do you know if this is still happening?

mithro, 2019-07-06:
@daveshah1 Please reopen if it is.

closed 2019-07-06 by mithro