tinyfpga/TinyFPGA-Bootloader #29: It fails to synthesize with latest toolchain.

elfmimi, 2018-10-04:
This is the error message it reports.
> fatal error: Top level port 'pin_usbp' assigned to an IO pad 'Y' and internal nodes
> Makefile:29: recipe for target 'bootloader.asc' failed

It seems to be a regression caused by this commit in arachne-pnr repository.
https://github.com/YosysHQ/arachne-pnr/commit/c00a14176e589f5ec4cb3914acc6683c608ac401

It completes successfully if you roll it back to previous commit. (commit id: 969836e)

Which side is responsible for fixing the issue?

mithro, 2019-07-06:
@elfmimi Please reopen if this still happening for you.

elfmimi, 2019-07-10:
Of course it still happens. Because the above PR referenced by microbuilder is not the one which aims to fix this. But it's the one which caused this. @mithro

mithro, 2019-07-11:
So, I'm leaning towards dropping support for arachne-pnr and only supporting nextpnr here.

elfmimi, 2019-07-12:
Sorry! My last report was wrong. I thought I had been pulling latest TinyFPGA-Bootloader, but I was failing to do so.
Both of anachne-pnr and nextpnr are processing latest source okay. > https://github.com/tinyfpga/TinyFPGA-Bootloader/commit/a77f828d3d6ae077e323ec96fc3925efab5aa9d7
Please clone this issue. @mithro

elfmimi, 2019-07-12:
My personal note.

First attempt to get rid of tristate inference was introduced on 2018-09-02
into smunaut/TinyFPGA-Bootloader:master at 
https://github.com/tinyfpga/TinyFPGA-Bootloader/commit/af173431f61796ac61f4095624bd0c5778e217a9
> boards/TinyFPGA_BX: Don't rely on tristate inference 

Merged to tinyfpga:master at 
https://github.com/tinyfpga/TinyFPGA-Bootloader/commit/858d1a92885f2e8c36933094827bc4f2b3af9a2b
with https://github.com/tinyfpga/TinyFPGA-Bootloader/pull/21

And is there until 2019-07-06
https://github.com/tinyfpga/TinyFPGA-Bootloader/blob/09b70464fff5a8c9db221c583b15e8a6ac843e9b/boards/TinyFPGA_BX/bootloader.v

But replaced by merge of osresearch/TinyFPGA-Bootloader:tomu on 2019-07-06 at
https://github.com/tinyfpga/TinyFPGA-Bootloader/commit/704c8fc96294d0cb70bdb2f873815de3e0281ae7
with https://github.com/tinyfpga/TinyFPGA-Bootloader/pull/34

Which is first introduced to osresearch/TinyFPGA-Bootloader:clock-crossing at
https://github.com/tinyfpga/TinyFPGA-Bootloader/commit/84959e7287d2f04c2f7bf3dcf15abb2750b24101
> TinyFPGA_BX: use nextpnr-ice40 
and merged to osresearch/TinyFPGA-Bootloader:tomu at
https://github.com/tinyfpga/TinyFPGA-Bootloader/commit/22f8d629520f5c1e2f7eb9ea93d98ebdbedd690c

mithro, 2019-07-12:
@elfmimi Thank you for confirming.

closed 2019-07-12 by mithro