tinyfpga/TinyFPGA-Bootloader #46: `tinyprog -l` fails on OSX

jeremyherbert, 2019-05-12:
When I run `tinyprog -l`, I get the following error:

```

    TinyProg CLI
    ------------
    Using device id 1d50:6130
    Only one board with active bootloader, using it.
    Boards with active bootloaders:

Traceback (most recent call last):
  File "/Users/jeremy/anaconda3/bin/tinyprog", line 11, in <module>
    sys.exit(main())
  File "/Users/jeremy/anaconda3/lib/python3.6/site-packages/tinyprog/__main__.py", line 298, in main
    print_board(port, m)
  File "/Users/jeremy/anaconda3/lib/python3.6/site-packages/tinyprog/__main__.py", line 196, in print_board
    print("        %s: %s %s" % (port, m[u"boardmeta"][u"name"], m[u"boardmeta"][u"hver"]))
  File "/Users/jeremy/anaconda3/lib/python3.6/site-packages/tinyprog/__init__.py", line 83, in __str__
    return "USB %d.%d" % (self.device.bus, self.device.port_number)
TypeError: %d format: a number is required, not NoneType
```

The problem is that on OSX (or at least on my high sierra install) `self.device.port_number` is `None`; in fact, the device is not enumerated as a serial port because there is no new serial port in /dev

https://github.com/tinyfpga/TinyFPGA-Bootloader/blob/99f87a5947fada082cc11606a2b3370f50e1f970/programmer/tinyprog/__init__.py#L141

mithro, 2019-05-15:
Can you test if the new master now works?

jeremyherbert, 2019-05-15:
Yep all good now. I'm gonna close this, please reopen if you need something else.

```
jeremy@mbp ~/code/TinyFPGA-Bootloader/programmer master $ tinyprog -l

    TinyProg CLI
    ------------
    Using device id 1d50:6130
    Only one board with active bootloader, using it.

    Boards with active bootloaders:

        USB 20.[no port number]: TinyFPGA BX 1.0.0
            UUID: fdf1d8e3-5c5e-405a-a2bc-54a7da2bd61a
            FPGA: ice40lp8k-cm81

jeremy@mbp ~/code/TinyFPGA-Bootloader/programmer master $ tinyprog --version
tinyprog 1.0.24.dev51+g6e31253 (b'tinyprog-1.0.23-51-g6e31253')
```

closed 2019-05-15 by jeremyherbert