tinyfpga/TinyFPGA-Bootloader #48: "TypeError: ord() expected string of length 1, but int found" - python3 (TinyFPGA BX, Ubuntu 19.04)

zrecore, 2019-05-13:
https://github.com/tinyfpga/TinyFPGA-Bootloader/blob/99f87a5947fada082cc11606a2b3370f50e1f970/programmer/tinyprog/__init__.py#L38

... Removing the call to ord() here allows the script to continue execution.

mithro, 2019-05-14:
@zrecore Can you please test this fixed your issue?

zrecore, 2019-05-14:
@mithro ok, testing now

zrecore, 2019-05-15:
Ok, no more python errors using `mithro/TinyFPGA-Bootloader` repo, `master` branch. 👍
(Still getting the gateware size issue, but that's probably a different problem)

[image]

mithro, 2019-05-15:
@zrecore It looks like your still failing? Can you log a full bug for that somewhere (maybe on https://github.com/timvideos/litex-buildenv/issues)?

zrecore, 2019-05-15:
@mithro added issue at https://github.com/timvideos/litex-buildenv/issues/137

PR #49 (2019-05-14): Make sure pretty_hex works under Python 2 and Python 3.
 * Run doctests under tox.
 * Add pretty_hex doctests for string, bytes and unicode

Fixes #48.
commit 1d9bfd78 (2019-05-14): Make sure pretty_hex works under Python 2 and Python 3.

 * Run doctests under tox.
 * Add pretty_hex doctests for string, bytes and unicode types.

Fixes #48
closed 2019-05-15 by zrecore