tinyfpga/TinyFPGA-Bootloader #50: hi i have a question ,why MUST have a  usbpull-up resistor

yangxingquan, 2019-05-29:


lolsborn, 2019-06-02:
So I'm not an expert so take this with a grain of salt, but it allows the bootloader to pull the USB_P line up indicating it's a "full speed" USB device.

```A USB device must indicate its speed by pulling either the D+ or D- line high to 3.3 volts. A full speed device, pictured below will use a pull up resistor attached to D+ to specify itself as a full speed device.```
From: https://www-user.tu-chemnitz.de/~heha/viewchm.php/hs/usb.chm/usb2.htm

It's tied to a pin on the FPGA instead of directly to 3.3v so your application can choose to enable it or not for your application.

yangxingquan, 2019-06-03:
@lolsborn  tks very much，may other reason is ,
No USB device, whether bus powered or self powered can drive the VBUS on its upstream facing port. If VBUS is lost, the
device has a lengthy 10 ms to remove power from the D+/D- pull-up resistors used for speed identification

closed 2019-07-06 by mithro