tinyfpga/TinyFPGA-Bootloader #51: Tests don't work -- missing files and no steps

xobs, 2019-05-30:
Currently, the tests do not work:

```
xobs@Pazuzu:~/Code/Fomu/TinyFPGA-Bootloader/tests/win10_enumeration_test$ make
iverilog -I.. -s top_tb -o test -c ../file_list.txt
../top_tb_header.vh:32: error: Unknown module type: vlog_tb_utils
../top_tb_header.vh:42: error: Unknown module type: vlog_tap_generator
3 error(s) during elaboration.
*** These modules were missing:
        vlog_tap_generator referenced 1 times.
        vlog_tb_utils referenced 1 times.
***
../test.mk:2: recipe for target 'test' failed
make: *** [test] Error 3
xobs@Pazuzu:~/Code/Fomu/TinyFPGA-Bootloader/tests/win10_enumeration_test$
```

They appear to depend on a module called `vlog_tb_utils`, which isn't mentioned anywhere.

If the files are added (e.g. from https://github.com/fusesoc/vlog_tb_utils), or if those lines are removed, then the error goes away, however the timestep never advances:

```
xobs@Pazuzu:~/Code/Fomu/TinyFPGA-Bootloader/tests/win10_enumeration_test$ make
iverilog -I.. -s top_tb -o test.vvp -c ../file_list.txt
vvp -v test.vvp
Compiling VVP ...
 ... VVP file version 11.0 (devel) (s20150603-616-ge93ee39)
Compile cleanup...
 ... Linking
 ... Removing symbol tables
 ... Compiletf functions
 ...      596 functors (net_fun pool=524288 bytes)
                148 logic
                  0 bufif
                  0 resolv
                220 signals
 ...      533 filters (net_fil pool=524288 bytes)
 ...     6375 opcodes (172032 bytes)
 ...      501 nets
 ...      596 vvp_nets (1048544 bytes)
 ...        0 arrays (0 words)
 ...       12 memories
                 12 logic (190 words)
                  0 real (0 words)
 ...       63 scopes
 ... 0.03125 seconds, 25184.0/5300.0/2296.0 KBytes size/rss/shared
Running ...
 ...execute EndOfCompile callbacks
 ...propagate initialization events
 ...execute StartOfSim callbacks
 ...run scheduler
           ... WAIT ~60 SECONDS ...
^C** VVP Stop(0) **
** Flushing output streams.
** Current simulation time is 0 ticks.
> finish
** Continue **
 ...execute Postsim callbacks
 ... 66.9531 seconds, 26844.0/5900.0/2888.0 KBytes size/rss/shared
Event counts:
           3 time steps (pool=113)
    38707312 thread schedule events
    232244005 assign events
             ...assign(vec4) pool=9362
             ...assign(vec8) pool=204
             ...assign(real) pool=256
             ...assign(word) pool=128
             ...assign(word/r) pool=204
         302 other events (pool=4096)

xobs@Pazuzu:~/Code/Fomu/TinyFPGA-Bootloader/tests/win10_enumeration_test$
```

rkapuscik, 2019-11-18:
I've tried running the tests as well and found they still don't work.
I added files from https://github.com/fusesoc/vlog_tb_utils manually, then used `-pfileline=1` flag in `iverilog`, turns out they are stuck in a loop over those lines:
```
../src/TinyFPGA-Bootloader_0/common/usb_fs_out_pe.v:205: Nonblocking assignment.
../src/TinyFPGA-Bootloader_0/common/usb_fs_out_pe.v:210: If statement.
../src/TinyFPGA-Bootloader_0/common/usb_fs_out_pe.v:211: Nonblocking assignment.
../src/TinyFPGA-Bootloader_0/common/usb_fs_out_pe.v:155: Event wait (@) statement.
../src/TinyFPGA-Bootloader_0/common/usb_fs_out_pe.v:157: Nonblocking assignment.
../src/TinyFPGA-Bootloader_0/common/usb_fs_out_pe.v:159: If statement.
../src/TinyFPGA-Bootloader_0/common/usb_fs_out_pe.v:163: Case statement.
```
Tests ran through `fusesoc` instead of manually adding files behave in the same way.
@tinyfpga could you take a look what could be wrong here?

kgugala, 2019-11-18:
this patch (https://pastebin.com/CjbzsUAV) solves the infinite loop issue, but the IP does not respond to USB packets in simulation.

PR #61 (2019-11-22): Simulation fixes
This PR fixes #51:
* adds missing files to both filelist and fusesoc core definition
* fixes simulation hangs by adding initial register values and switching to blocking assignments in combinational logic generation.

However, I see some errors in the simulation:

```
fusesoc --cores-root `pwd` run --run --target win10_enumeration_test TinyFPGA-Bootloader --vcd --timescale="1ps / 1ps" --continue_on_fail
INFO: Preparing ::vlog_tb_utils:1.1                                                                                                                                              [... cut]
closed 2019-11-22 by mithro