tinyfpga/TinyFPGA-Bootloader #55: Add retiring of sector writes if they fail

mithro, 2019-07-06:
As @ewenmcneill suggested.

ewenmcneill, 2019-07-06:
These two commits from an old PR give the general code structure, but probably need reimplementing to fit with the much rearranged code in master now:

*   https://github.com/tinyfpga/TinyFPGA-Bootloader/pull/16/commits/9b71b914efc330498c414dc8b443e42b8087aef5
*    https://github.com/tinyfpga/TinyFPGA-Bootloader/pull/16/commits/6798b2a2626ae6c3ccca66e39bdf8320e1bf75e1

Ewen

ewenmcneill, 2019-07-12:
FTR, another user having problems, supposedly with the "latest" tinyprog (installed from git).  I've asked whether they installed it before or after the changes merged this past weekend (2019-07-12).  But either way, retrying the failed sectors might have helped them.

https://discourse.tinyfpga.com/t/issues-with-tinyprog-u-on-bx/628/18

(and this is another reason for wanting to actually do a release to PyPI...)

Ewen

ewenmcneill, 2019-07-15:
> FTR, another user having problems, supposedly with the "latest" tinyprog (installed from git). I've asked whether they installed it before or after the changes merged this past weekend (2019-07-12).

Apparently they installed just *before* the fix was merged, and it's been working better for them after updating to the current installed version -- https://discourse.tinyfpga.com/t/issues-with-tinyprog-u-on-bx/628/20.

They also suggest that the PyPI version be updated... which I also agree would be a good idea.

Ewen